SolidGeek/VescUart #1: Compatibility with SoftwareSerial port

Peemouse, 2018-10-30:
Hi,

Is it possible to use the library with SoftwareSerial port ?
I did a quick test, it does not compile (didn't dive further).

Could you point out what need to be done or could you add this compatibility ?

Thanks a lot.

Clément

CTSchorsch, 2018-11-18:
Yes, i would appreciate this Feature too

SolidGeek, 2018-11-18:
I am not sure, however, I made the Serial port for debugging (debugPort) compatible with SoftwareSerial by defining it as the parent class Stream. 

`Stream* debugPort = NULL;`

I do not have the time to try this out myself at the moment, but if you change all instances of HardwareSerial with Stream, I think you can at least compile. Could you try this out? 

Like so: `Stream* serialPort = NULL;`

CTSchorsch, 2018-11-19:
Ok, found my misstake. The VESC sends 3.3V, its to low for the arduino leonardo on pin 12/13.

SolidGeek, 2018-11-19:
Glad to know it works at some level. Have you tried using other pins (I do not have any experience with the Leonardo), or maybe another Arduino? Would be interesting to see if it works reliably. If you make it work I will update the library, and add an example for others to use 👍

CTSchorsch, 2018-11-20:
Its working ! I tried different pins on the arduino uno. everything works fine with this :)

but, i tried to start the motor with setDuty() and the motor is running only for a second. is this not the right command to start and stop the motor ?

Peemouse, 2018-11-20:
@CTSchorsch did you set the VESC up with "UART" only in "App Settings - General - App to use" ?

CTSchorsch, 2018-11-20:
Yes, this mistake i found by myself :)
It is funny, with the vesc-tool it works fine and i think it uses uart too

SolidGeek, 2018-11-20:
It is not enough to send the duty cycle ones, you will have to keep sending otherwise the VESC will timeout and stop any motion. Do you send the command continuously?

CTSchorsch, 2018-11-20:
No, thanks, thats it !
But i run in the next problem :) The Uno has too less memory to handle VESCUart and SD Card. I ordered a Zero for further testing. Thanks a lot, i think this issue is done :)

closed 2019-03-06 by SolidGeek

RFRIEDM, 2019-05-23:
What do I need to change to use software serial on an Uno. I just pulled master today from here. 

So far, I have changed VescUart.h to have the line:

`Stream* debugPort = NULL;`

And, getVescValues.ino as follows, naming Serial1 to a softwareserial port called serial1.

```c
#include <buffer.h>
#include <crc.h>
#include <datatypes.h>
#include <VescUart.h>

/*
  Name:    getVescValues.ino
  Created: 19-08-2018
  Author:  SolidGeek
  Description:  This example is made using a Arduino Micro (Atmega32u4) that has a HardwareSerial port (Serial1) seperated from the Serial port. 
                A Arduino Nano or Uno that only has one Serial port will not be able to display the data returned.
*/

#include <VescUart.h>
#include <SoftwareSerial.h>

/** Initiate VescUart class */
VescUart UART;

//Create software serial port
SoftwareSerial serial1(10, 11); // RX, TX

void setup() {

  /** Setup Serial port to display data */
  Serial.begin(9600);

  /** Setup UART port (Serial1 on Atmega32u4) */
  serial1.begin(19200);
  
  while (!Serial) {;}

  /** Define which ports to use as UART */
  UART.setSerialPort(&serial1);
}

void loop() {
  
  /** Call the function getVescValues() to acquire data from VESC */
  if ( UART.getVescValues() ) {

    Serial.println(UART.data.rpm);
    Serial.println(UART.data.inpVoltage);
    Serial.println(UART.data.ampHours);
    Serial.println(UART.data.tachometerAbs);

  }
  else
  {
    Serial.println("Failed to get data!");
  }

  delay(50);
}
```

marcel007, 2020-02-13:
I am trying to use SoftwareSerial via Pins 50,51 and 52,53 for communicating between a ATMEGA 2560 and two VESCs (two rear wheels) via defined serial4 and serial5 . I tested the above shown program at first and the compiler of Arduino IDE 1.8.10 returns an error as follows:

VESC:34:30: error: no matching function for call to 'VescUart::setSerialPort(SoftwareSerial*)'

   UART.setSerialPort(&serial1);

                              ^

In file included from C:\Users\Marcel\Downloads\VESC\VESC.ino:4:0:

C:\Users\Marcel\Documents\Arduino\libraries\VescUart-master\src/VescUart.h:49:8: note: candidate: void VescUart::setSerialPort(HardwareSerial*)

   void setSerialPort(HardwareSerial* port);

        ^~~~~~~~~~~~~

C:\Users\Marcel\Documents\Arduino\libraries\VescUart-master\src/VescUart.h:49:8: note:   no known conversion for argument 1 from 'SoftwareSerial*' to 'HardwareSerial*'

exit status 1
no matching function for call to 'VescUart::setSerialPort(SoftwareSerial*)'

Do I need to change VescUart.h?

AmanuelEphrem, 2020-07-20:
Yes, you need to change VescUart.h and VescUart.cpp. Wherever you see hardwareserial* change to softwareserial*

dpbnasika, 2020-12-28:
> Its working ! I tried different pins on the arduino uno. everything works fine with this :)
> 
> but, i tried to start the motor with setDuty() and the motor is running only for a second. is this not the right command to start and stop the motor ?

Hello, How did you manage to get the Vesc values such as rpm, voltage etc and and print on arduino serial terminal?,  when I run the same code, it does not receive the values on rx. Do you know why this is happening?.

CTSchorsch, 2020-12-28:
> Hello, How did you manage to get the Vesc values such as rpm, voltage etc and and print on arduino serial terminal?,  when I run the same code, it does not receive the values on rx. Do you know why [...]

Hi, can you post your code ? You need one serial line to communicate with the VESC and one to communicate with the PC. Double check pins and wireing

AmanuelEphrem, 2020-12-28:
> Hello, How did you manage to get the Vesc values such as rpm, voltage etc and and print on arduino serial terminal?, when I run the same code, it does not receive the values on rx. Do you know why [...]

I second what @CTSchorsch had to say. Also make sure you have a common ground, as that mistake is what caused me to not receive any data when I first started using it.

dpbnasika, 2020-12-30:
Hi CTSchorsch & Amanuel, 

Thank you for your quick responses,
 
I used Arduino mega2560 board and I connected the rx and tx pins of VESC to 18 and 19, and used the same example code as below:

#include <VescUart.h>

/** Initiate VescUart class */
VescUart UART;

void setup() {

  /** Setup Serial port to display data */
  Serial.begin(9600);

  /** Setup UART port (Serial1 on Atmega32u4) */
  Serial1.begin(19200);
  
  while (!Serial) {;}

  /** Define which ports to use as UART */
  UART.setSerialPort(&Serial1);
}

void loop() {

  /** Call the function getVescValues() to acquire data from VESC */
  if ( UART.getVescValues() ) {

    Serial.println(UART.data.rpm);
    Serial.println(UART.data.inpVoltage);
    Serial.println(UART.data.ampHours);
    Serial.println(UART.data.tachometerAbs);

  }
  else
  {
    Serial.println("Failed to get data!");
  }

  delay(50);
}

**I changed the baud rate to 115200 and then it started working.  I guess because Vesc is also sending the data at 115200.**

I have another question: 

is it possible to control the position (0-360 degrees) of the bldc motor using VESC only with hall sensors in FOC mode?. I have seen some methods like setPosition in 
https://github.com/RollingGecko/VescUartControl/blob/master/VescUart.h

Calling the method through the Arduino code would set the position to desired degrees with out damaging the motor or sensors or VESC ?.

Did any one tried that?.

AmanuelEphrem, 2020-12-30:
I have never tried controlling a motor with a VESC using an arduino, but you should be able to use the implementation from RollingGecko. I forked the project and added the implementation, but I haven't tested it. Feel free to download my forked version and let me know if there are any bugs! As for safety, it **_should_** be completely safe changing the position using the library, but I have never personally done it. I would always change the degrees to always be on the interval [0,360] though (for example I would change 540 degrees to be 180 degrees). Hope this helps.

dpbnasika, 2020-12-31:
> I have never tried controlling a motor with a VESC using an arduino, but you should be able to use the implementation from RollingGecko. I forked the project and added the implementation, but I [...]

Hi, I too had the changes already in the library, but I din't test it and would like to see if any one had controlled it safely by using that method.