SolidGeek/VescUart #2: Features

CTSchorsch, 2018-11-27:
Hi,

your lib works really fine. Thank you for your work. 

Do you think it is possible to read the Temperature and the drv error too?

Greetings 
Georg

SolidGeek, 2018-11-28:
Yeah that should be quite easy. I am actually skipping the first 4 bytes of the data received, which contains the mosfet temperature (2 bytes) and the motor temperature (2 bytes). Regarding the error you are looking for, I can get you the mc_fault_code, which can hold any of these values:

`typedef enum {
	FAULT_CODE_NONE = 0,
	FAULT_CODE_OVER_VOLTAGE,
	FAULT_CODE_UNDER_VOLTAGE,
	FAULT_CODE_DRV,
	FAULT_CODE_ABS_OVER_CURRENT,
	FAULT_CODE_OVER_TEMP_FET,
	FAULT_CODE_OVER_TEMP_MOTOR
} mc_fault_code;`

I can implement it in a dev branch if you want to try it out?

CTSchorsch, 2018-11-28:
Hi,

yes. This would be fine. I use your lib to write the housekeeping data to a sdcard and this values are missing :)

Georg

CTSchorsch, 2018-11-29:
> Made a new branch with the changes. You can see the changes here:
> 87a2552
>
> And download the updated library here:
> https://github.com/SolidGeek/VescUart/tree/development
>
> Let me know if it works 👍

some error in your lib

error: invalid conversion from 'uint8_t {aka unsigned char}' to
'mc_fault_code' [-fpermissive]
    data.error     = message[ind];
                                ^
exit status 1

SolidGeek, 2018-11-29:
I have changed the datatype of the error variable to uint8_t, should work now. Its weird, because I didn't get the error on compiling. Would you give it a try again? Just redownload the development branch 👍

CTSchorsch, 2018-11-30:
Yes, thanks, now its working (compiling and values are fine)
The error value is one from the mc_fault_code ENUM ?

SolidGeek, 2018-11-30:
Glad to hear. Yeah the error value map directly to the mc_fault_code ENUM. So an error value of 3 would mean a DVR error etc.

CTSchorsch, 2018-11-30:
Perfect :)

vicgwee, 2019-08-21:
Is there a possibility to reduce the time taken by getVescValues()? 

One option is to increase the baud rate. On my Mega, I've recorded the following times taken (averaged over 10,000 tries): 

Baud Rate | Time taken (ms) 
-- | -- 
9600 | 79.04  
40000 | 19.30 
115200 | 6.98 
200000 | 4.23 
250000 | 3.46   

*Over 400,000 I begin to encounter some communication failures.

Wondering what's the bottleneck: unpacking the payload? or receiving the payload from VESC.

What if we could separate the function into two: 
getVescValues() -- send the request to the VESC
checkVescValues() -- check the Serial buffer if the payload has been received, if so then unpack the payload and update the values of .data

Thus the code doesn't block the arduino while waiting to receive the payload.

daniel84uk, 2019-08-26:
Thanks for this, i spent ages trying to figure out how to read the bloody temps lol.