SolidGeek/VescUart #4: Doesn't connect to STM32 Blue Pill over Serial1 (PA9 & PA10)

neilbalch, 2019-03-18:
I've been trying to get the sample code to read status parameters to work with my FSESC 4.12 over the STM32 Blue Pill's USART1 port _(PA9 and PA10)_. No such luck, as I keep getting `Failed to get data!` messages. Do you have any idea what could be causing this?

SolidGeek, 2019-03-18:
I would appreciate some more information, otherwise, it is very hard for me to diagnostic. 

- Do you use the Arduino core for STM32?
- Have you updated to the latest VESC firmware (and using the development branch)?
- Please supply some sample code

neilbalch, 2019-03-18:
Sure! I'm using the Arduino core:

```
[env:genericSTM32F103C8]
platform = ststm32
framework = arduino
board = genericSTM32F103C8
upload_protocol = dfu
monitor_speed = 250000
```

Yes, I have updated to the latest firmware version with the updater tool. What is the development branch?

The sample code is yours, the `getVescValues` example.

SolidGeek, 2019-03-19:
You should use the development branch indstead of the master, if you use the newest VESC firmware. 

Have you tried one of the other Serial ports?

neilbalch, 2019-03-20:
1. How would I do that with PlatformIO? I can only grab libraries straight from the master branch.
2. Yes, I've tried all three of the USART ports on my Blue Pill board.

oiitsjamesmate, 2019-03-20:
Small world, I too am trying to get vesc data on an STM32 and mine worked just fine. Are you using Serial1.begin( *insert baud rate here* ); on PA9 and PA10? Then its the exact same as the test sketches. Working on the current development branch. I have attached the zip file containing the development branch. Hope it works!
[VescUart-development.zip](https://github.com/SolidGeek/VescUart/files/2988095/VescUart-development.zip)

neilbalch, 2019-03-21:
You described the hardware setup perfectly. I'm still not clear how I should be using the development branch in PlatformIO. As far as I know, it only supports using the master branch.

neilbalch, 2019-03-23:
By the way, I still can't connect to the VESC, even when using the development branch.

```
#include <Arduino.h>
#include <HardwareSerial.h>
#include <VescUart.h>

#define VESC_USART Serial1

VescUart UART;

void setup() {
  Serial.begin(9600);
  VESC_USART.begin(19200);
  // while (!Serial) {;}
  UART.setSerialPort(&VESC_USART);
}

void loop() {
  if (UART.getVescValues() ) {
    Serial.println(UART.data.rpm);
    Serial.println(UART.data.inpVoltage);
    Serial.println(UART.data.ampHours);
    Serial.println(UART.data.tachometerAbs);
  } else {
    Serial.println((String)millis() + " Failed to get data!");
  }

  delay(50);
}
```

neilbalch, 2019-03-30:
Deeeeeeerp. The example has the baud rate set at `19200` bps while the VESC communicates at `115200` bps by default. I forgot to check. Now it works properly with my code communicating with the VESC at `115200` bps.

closed 2019-03-30 by neilbalch