SolidGeek/VescUart #5: Problem to compile

oliverSKP, 2019-03-20:
Hello,
Firstly, thank you for your library. 
However I have a problem compiling your example getVescValue for an arduino due, I use your develepment branch, I have a vesc fw3.51 (hw4.12). 

 I have the following error message: 

exit status 1
request for member 'setSerialPort' in '1074661376u', which is of pointer type 'Uart*' (maybe you meant to use '->' ?)

After several hours of research, I cannot find why it does not compile, do you have any solution?

oliverSKP, 2019-03-22:
Problem solved, I had to rename the object Uart : VescUart Uart into VescUart Uart2.

closed 2019-03-22 by oliverSKP

aligoodman, 2021-09-03:
@oliverSKP , how do you do this sorry, I am having the same issue? Thanks