SolidGeek/VescUart #6: Work with VESC 6

pumpkinheadnick, 2019-05-12:
Hello. 

I have a VESC 6 and I can't get it to read any data. I have the UART wired to the Serial1 of an arduino mega. 
Any ideas why it would not work? Let me know if you need more info.

RFRIEDM, 2019-05-31:
1. Cross TX and RX from Serial1 to the VESC
2. Ensure your baud rates match. Although I can set the baud rate in VESC Tools to other baud rates, my ESC always seems to use 115200, so that's what I run the Arduino at. If you are unsure, try all the common baud rates:)
3. Check a good ground between Arduino and ESC. A scope helped me debug data was going in both directions.

gianmarcov, 2019-11-14:
Hi, RFFRIEDM, this repo is very messy and does not full implement vesc interface.
i invite yout to use my new implementation https://github.com/gianmarcov/arduino_vesc it cover all the functionallity of the vesc.

It's also well documented https://gianmarcov.github.io/arduino_vesc

closed 2020-02-15 by SolidGeek