SolidGeek/VescUart #7: How to use COMM_FORWARD_CAN to send commands to slave VESC

joseguerr, 2019-06-21:
Hi there @SolidGeek 

Have been searching for some code that has this function implemented so I can use it in the arduino, but so far without success. I don't know what are the transformations required in the messages to actually send such command, using the libraries that are currently created.

The idea here was to sent commands to a slave vesc, for instance, setting RPM's. I've tested the CAN master-slave communication through VESC tool and is working well. It is just the code adaptation in c++ that I have no idea how to perform. I don't even know how is supposed to work with such function, while sending a command.

I would appreciate the help of anyone who reads this. Have a nice day!

SolidGeek, 2020-02-15:
Hi joseguerr, sorry for the late reply. I have not much experience in the CAN communication between master and slave VESC, so, unfortunately, I cannot help with this at the moment. Do you have any documentation/examples of the usage of COMM_FORWARD_CAN?

loicalini, 2020-07-06:
https://github.com/astulnikov/ESk8Arduino/blob/405fe467cace6ee455dba3c70ad3bc34eb63c2aa/lib/VescUart/src/VescUart.h

^check out this similar library that has CAN forwarding implemented

SystemDiagnosticss, 2021-08-06:
Any updates in this case?

SystemDiagnosticss, 2021-08-06:
@loicalini why do you think thats in this fork CAN forwarding is implemented?

SolidGeek, 2022-12-18:
This was implemented by a pull request 👍

closed 2022-12-18 by SolidGeek