SolidGeek/VescUart #10: Oscillations with setDuty

SwarleyAUS, 2019-08-19:
Hey this might not be an issue with the code but I am having a problem that I am hoping you can help with. I can read and send commands (somewhat) fine using the library. However, if the duty cycle changes too quickly when returning to 0 (what I am passing in via setDuty), I get these crazy oscillations from -94 to +94% throttle in the VESC which is unrecoverable aside from controller reboot. Any ideas? Duty is a number between -1.0 and 1.0. 

[image]

Same thing happens if I do, for example:

Vesc.setDuty(0.25);
delay(1000);
Vesc.setDuty(0.5);
delay(1000);
Vesc.setDuty(0.0);

If I slowly bring the control loop back to 0% throttle I don't have the issue.

Thanks!

SolidGeek, 2019-08-19:
I have never used the duty cycle control method before, as it to me has always been acting kind of weird. But please supply me with full code example, and which version of the BLDC firmware and what version of VescUart you are using 👍  To see if I can replicate the issue.

SwarleyAUS, 2019-08-20:
Thanks for your reply. I think I *may* have fixed it... it seemed to be due to me calling setDuty at one frequency and then calling getVescValues at a different frequency and the timing around that. I believe because I was doing if (millis - VescTime) >= 10) {setDuty();} at one point in the loop and if (millis - VescTime2) >= 100) {getVescValues();} at another it was confusing the VESC.

This is my latest code that seems to be working, as in if thrValue abruptly goes from a number >>127 or <<127 back to 127 (0% duty), it doesn't glitch: https://github.com/SwarleyAUS/ESP32_1Wheel/blob/master/ESP_CONTROLLER.ino -> making a DIY onewheel

So my code as I expected =) This is with HW 4.12, FW 3.50 (I think).

Cheers,
Taylor

closed 2019-08-20 by SwarleyAUS