SolidGeek/VescUart #18: Power reading and writing

Aurel-MT, 2020-06-30:
Hello, thank you for the great library!
Is it possible to get a momentary power reading from the VESC?
I tried to use the inpVoltage multiplied by the avgImputCurrent but this just gives me an average reading. 
My second question is if there is a way to apply a certain power to the Motor regardless of the RPM?
If I use setCurrent function the power increases with higher RPM.

SolidGeek, 2020-07-02:
Hey, the avgImputCurrent is calculated by the VESC each time you request data. Thus, if you request data with a period of 500ms, the current measurement is average over the last 500ms. So as long as you sample fast enough it should be just fine to calculate the wattage. 

Regarding setting a certain motor power, I don't see why the power should increase with the RPM when you use setCurrent()? If you set a motor current of 1A and has an input voltage of 42V, you will get 42W. The motor simply runs as fast as it can at that current. If you want to control in terms of watt, you could simply use the measured input voltage to find the current needed for a wanted wattage.

Aurel-MT, 2020-07-03:
Hey, thank you for your fast response!
I sample about every 2ms so it should be perfect to calculate the wattage.
I think the VESC increases the motor voltage according to the RPM, so the power is changing even if setCurrent() stays the same. 

Because I wasn’t sure I did a small test: 
Voltage is about 45V
UART.setCurrent(2.0)

And then printed out:
UART.data.rpm
UART.data.avgMotorCurrent       // Is the same as setCurrent 2A
UART.data.avgImputCurrent
UART.data.avgImputCurrent*UART.data.impVoltage     // Power

The Power increased with higher rpm, like shown in the picture. 
Have I done something wrong ?

[image]

SolidGeek, 2020-11-06:
You do not show timestamp or voltage in the table, however I do believe that is an effect of the motor spinning up. It has some kind of buildin ramp, to steadily increase the current. The avgMotorCurrent may be a setpoint, and not a measured value since it perfectly matches 2 amps. I am closing this issue, as it does not seem to be a problem with the library.

closed 2020-11-06 by SolidGeek