SolidGeek/VescUart #26: Set minimum speed while dealing with obstacles

idoadse, 2021-05-12:
We are building a pushcart using 2 HUB motors, VESC, and Arduino to control the VESC using U-ART (Rolling Gecko)
used the "SetRPM" function to control my speed with fast and slow toggle button to switch speeds.

We encounter 2 issues:
1. setRPM will "slow me" down if I push the cart to move faster (I'd like to avoid it)
2. In low speed, the cart has no enough power to overcome obstacles. 
It seems like setRPM for high values, and limiting the ERPM to the requested speed will bring more power to the engine  and help overcome this issue - i.e. use the ERPM to control the speed and keep "setRPM" on high values. how ever I'm not able to toggle speeds in this configuration, due to the lack of option to dynamically control the ERPM.

- We have also posted in Vesc forums: https://vesc-project.com/node/3011

- Do you have some other solutions? 
- Is there an option to somehow tweek the values of erpm from the arduino code?

Thanks

vicgwee, 2021-06-15:
Hope this reply doesn't come too late. The issue is with configuring your VESC properly. In the VESC Tool, these settings are located in the PID Controllers section.

1. There is a setting "Allow Braking". Set it to False to avoid the issue you faced. 

2 There is a setting "Minimum ERPM". Below this ERPM, there will be no power output to the motors. I have not tried setting this to 0, but I have reduced it from the default of 900 to 100 successfully. To control it at low speeds, I use duty cycle control `setDuty()`. This will not get you as accurate a speed reading, but it is sufficient for my purposes.

idoadse, 2021-06-16:
Thanks for the reply! 
1. I will test the "Allow Braking" sound exactly what I need
2. I have indeed work around it with setDuty(). If anyone get similar issue, my approach was to identify when rpm is small from some threshold (meaning it encountered obstacle). In this case I exponentially increase rpm until rpm is back to normal

SolidGeek, 2022-12-18:
This is not an issue with the library, but rather a discussion of how to use the library. I hope you found a solution to your problem :-)

closed 2022-12-18 by SolidGeek