SolidGeek/VescUart #28: Feature Request

h2omatt, 2021-06-06:
First off I am new to the VESC community and this library worked first try and is really awesome. 

I don't know if this is the best place to ask but I really need help. I would like to be able to use setRPM to control the motor speed but also be able to limit the motor current. I have tried to do speed control by using a PID loop using setCurrent and then UART.data.rpm and my PID parameter. 

Is there a way I could send a setRPM and setMaxCurrent command? I am open to suggestions too.

SolidGeek, 2021-06-06:
Hi mate, glad you like the library!

There is no command for setting the max current in the COMM_PACKET_ID enum data structure. 
Instead one has to read the entire configuration using COMM_GET_MCCONF, change the max current and then update the settings using COMM_SET_MCCONF. This is not meant to be done live, meaning, that the configuration should only be changed during startup or at standstill. I don't know if this could be useful? 

Did you manage to get the PID control working? I have not tried myself, but I figure, that the RPM has to be read very fast, so increasing the baud-rate and sample-rate might be helpful.

h2omatt, 2021-06-13:
SolidGeek, thanks for the response!

I did get the PID working! it works amazing actually. I had a very dumb programming error but once fixed it works perfect.

closed 2021-08-17 by h2omatt