SolidGeek/VescUart #35: Error declaration of 'foc_hfi_samples and extra qualification on sendKeepalive

Trey-Raventor, 2022-05-10:
I am using the setCurrent example and getting errors on the current master. The previous version did not give these errors and it appears the sendKeepalive was not a function in the previous version. I appreciate the work adding canID going forward. It will be very helpful for dual VESC control.

Errors with directory information stripped:

libraries/VescUart/src/VescUart.h:5:0,
                 from /tmp/.arduinoIDE-unsaved2022410-33799-j9cdvk.tgqpq/sketch_may10a/sketch_may10a.ino:1:
libraries/VescUart/src/datatypes.h:394:18: error: declaration of 'foc_hfi_samples <anonymous struct>::foc_hfi_samples' [-fpermissive]
  foc_hfi_samples foc_hfi_samples;
                  ^
libraries/VescUart/src/datatypes.h:240:3: error: changes meaning of 'foc_hfi_samples' from 'typedef enum foc_hfi_samples foc_hfi_samples' [-fpermissive]
 } foc_hfi_samples;
   ^
/tmp/.arduinoIDE-unsaved2022410-33799-j9cdvk.tgqpq/sketch_may10a/sketch_may10a.ino:1:0:
/home/rhiannon/Desktop/SBC Project/Arduino Code/libraries/VescUart/src/VescUart.h:171:14: error: extra qualification 'VescUart::' on member 'sendKeepalive' [-fpermissive]
         void VescUart::sendKeepalive(void);
              ^
libraries/VescUart/src/VescUart.h:177:14: error: extra qualification 'VescUart::' on member 'sendKeepalive' [-fpermissive]
         void VescUart::sendKeepalive(uint8_t canId);
              ^

Compilation error: exit status 1

Trey-Raventor, 2022-05-10:
Just checked some other Arduino boards. The errors are showing for the Due, but not showing for the Mega.

matthewryanburgin191, 2022-05-23:
This is also occuring on esp32 boards for me, is there a work around?

SolidGeek, 2022-05-25:
I will take a look into this. @Trey-Raventor did you find a solution?

matthewryanburgin191, 2022-05-27:
@SolidGeek @Trey-Raventor compiles on esp32. Thank you

Trey-Raventor, 2022-05-27:
Thank yall for looking into this. I will check the latest code when I get
back from my trip.

What I have done to get by is use the previous version of code (before CAN
ID) and only update the CAN ID functions for set current and set brake
current. This was all I needed for my current project.

Trey-Raventor, 2022-05-31:
The latest code compiles without errors for me as well. I will do some more in-depth tests later this week, but thank you for updating and maintaining this great library.

closed 2022-05-31 by Trey-Raventor