SolidGeek/VescUart #40: Integration with esphome

bogdanr, 2022-12-16:
I am trying to integrate this with esphome.

My config is as follows:
```
substitutions:
  id: test3d1mini32
  project: Test D1 Mini32
esphome:
  name: $id
  libraries:
    - solidgeek/VescUart
esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino
    version: 2.0.4
```

I receive the following error when I try to compile this:
```
In file included from /data/test3d1mini32/.piolibdeps/test3d1mini32/VescUart/src/VescUart.cpp:2:
/data/test3d1mini32/.piolibdeps/test3d1mini32/VescUart/src/VescUart.h:27:9: error: redeclaration of 'float VescUart::dataPackage::tempMotor'
   float tempMotor;
         ^~~~~~~~~
/data/test3d1mini32/.piolibdeps/test3d1mini32/VescUart/src/VescUart.h:14:9: note: previous declaration 'float VescUart::dataPackage::tempMotor'
   float tempMotor;
         ^~~~~~~~~
/data/test3d1mini32/.piolibdeps/test3d1mini32/VescUart/src/VescUart.cpp: In member function 'bool VescUart::processReadPacket(uint8_t*)':
/data/test3d1mini32/.piolibdeps/test3d1mini32/VescUart/src/VescUart.cpp:219:3: error: jump to case label [-fpermissive]
   default:
   ^~~~~~~
/data/test3d1mini32/.piolibdeps/test3d1mini32/VescUart/src/VescUart.cpp:217:13: note:   crosses initialization of 'uint32_t mask'
    uint32_t mask = 0xFFFFFFFF;
             ^~~~
/data/test3d1mini32/.piolibdeps/test3d1mini32/VescUart/src/VescUart.cpp:223:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
*** [/data/test3d1mini32/.pioenvs/test3d1mini32/libd76/VescUart/VescUart.cpp.o] Error 1
```

How can I work around this?
What version of Arduino is it compatible with?

SolidGeek, 2022-12-18:
I might have introduced a compile error in a previous pull request. I have updated the development branch, could you please give it a try?: [VescUart/tree/development](https://github.com/SolidGeek/VescUart/tree/development)

bogdanr, 2023-12-09:
It compiles fine with the code from the development branch.