SolidGeek/VescUart #44: Vesc

JavierDebon, 2023-05-26:
I am trying to make a display with arduino MEGA using the SolidGeek library but I am not getting good results. Could you tell me how to make the connections and how to do it thanks.

SolidGeek, 2023-05-27:
You need to connect a serial port from your Arduino Mega, to the serial port on the VESC. RX goes to TX, and TX goes to RX. Remember to connect GND also. Then it should be a matter of selecting the correct Serial port, using the `setSerialPort()` function.

JavierDebon, 2023-06-03:
thank you very much it was my mistake I had a broken cable. Thank you.
I have another question I would like to know the independent temperature of each mosfet. In your code you do `data.tempMosfet = buffer_get_get_float16(message, 10.0, &index); // 2 bytes - mc_interface_temp_fet_filtered()` but I don't know which mosfet that temperature is from. could you help me?